STMicroelectronics/STM32Cube_MCU_Overall_Offer #1: i-cube_lrwan repo?

sslupsky, 2019-09-21:
Hello,

Do you plan to provide a github repo for the i-cube_lrwan library?

https://www.st.com/en/embedded-software/i-cube-lrwan.html

Kind regards,
Steve

josesimoes, 2019-11-18:
yes, please

CCASTM, 2020-01-20:
Hello Steve and all,
Sorry, I missed this Issue, as it does not correspond to an effective project.
For the moment we concentrate on the STM32 Fw Packages. From now on, we will publish individual FW Components projects (CMSIS and CMSIS Devices have been pubished)
Next ones will be the HAL Drivers, then MW and so on
About the Cube Expansion Packages (the one you mentioned), we have no plan to communicate yet. We , as ST R&D division, are mainly in charge of ''base'' projects, the Cube Expansion are proposed by other ST R&D Division.
I recommend to contact the people having posted the i-cube-lrwan projects to get more information about their intention regarding GitHub
Regards
Christophe

closed 2021-02-15 by CCASTM

awneil, 2021-07-06:
> I recommend to contact the people having posted the i-cube-lrwan projects to get more information about their intention regarding GitHub

So who, exactly, would be the right people to contact about this ?

The stated aim of this repo is:

> STM32Cube is an STMicroelectronics original initiative to ease the developers life by reducing efforts, time and cost.

That aim would certainly be furthered greatly if it really were a central point for **_all_** ST content ...