STMicroelectronics/STM32Cube_MCU_Overall_Offer #2: Missing (PCD_HandleTypeDef *) casts cause build errors in C++ compilers when USB is used

pawelszramowski, 2020-06-06:
Since `void*` is not implicitly convertible to other pointers, missing `(PCD_HandleTypeDef *)` casts cause build errors in C++ compiler (I used IAR EW for Arm 8.50.1) when USB is used. This affects the following files:
* `Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c` included in firmware packages downloaded by STM32CubeMX and copied to or referenced by user projects. Replace `PCD_HandleTypeDef *hpcd = pdev->pData;` with `PCD_HandleTypeDef *hpcd = (PCD_HandleTypeDef *)pdev->pData;` to fix it.
* `usbd_conf.c` generated from `db/templates/usbdconf*_c.ftl` templates in STM32CubeMX installation directory. Replace `(HAL_PCD_.*\()(pdev->pData)` with `$1\(PCD_HandleTypeDef \*\)$2` using regular expressions to fix it in `HAL_PCD_*` function calls.

CCASTM, 2021-02-15:
Ali,
I have discovered this Issues, quite old !
Can you check in a first step with Aymen if this problem has already been taken into account ?
Regards
Christophe

ALABSTM, 2021-02-15:
Hi @pawelszramowski,

Thank you for your report. May I know the MCU target you are using? This way, the issue will be transferred to the appropriate repository and deeper analyzed.

With regards,

pawelszramowski, 2021-02-15:
@ALABSTM, this is not related to a particular microcontroller. As I wrote, it affects files `%ProgramFiles%\STMicroelectronics\STM32Cube\STM32CubeMX\db\templates\usbdconf*_c.ftl` and `%UserProfile%\STM32Cube\Repository\STM32Cube_FW_F3_V1.11.0\Middlewares\ST\STM32_USB_Device_Library\Class\CDC\Src\usbd_cdc.c` thus project for any MCU generated using these files will be affected too.

ALABSTM, 2021-02-17:
Hi @pawelszramowski,

Thank you for the precision. Indeed, the issue might impact more than one firmware in particular. I asked my question simply because with the way-of-work we defined, such issues are handled at `STM32Cube..` repository-level, not at this one's.

Once the issue confirmed and logged into our internal database, we will surely spread the fix to any other firmware impacted. I noticed you have mentioned the `STM32Cube_FW_F3_V1.11.0`. I will then transfer this issue to the `STM32CubeF3 repository`.

Thank you for your comprehension and thank you again for having reported the point.

With regards,