STMicroelectronics/STM32Cube_MCU_Overall_Offer #3: SVD Files

svcguy, 2021-01-16:
One thing that would be great is if ST could have *.svd files for all device families on GitHub.  If linked here as well, this could be a one-stop shop to have everything needed source-wise to develop and debug applications on STM32 hardware.

CCASTM, 2021-02-15:
Hello svcguy
Sorry I missed this Issues for a while. We do not receive many customers comments or questions in this repo.
Your proposal sounds good. I understand that you propose to create dedicated repos containing the SVD files (one repo per STM32 series for example) I will launch a formal request internally and come back to you
Regards
Christophe

CCASTM, 2021-02-15:
Internal ST Reference 101395

loicfrance, 2021-03-06:
could it be possible to have in the same repository the mapping [chip name (e.g. STM32F303K8)] --> [svd file] ?

PierreLeCorre, 2022-02-21:
Dear all,
_This issue is open since long time, sorry for this._
Following internal discussion, we do not plan to publish the SVD files associated to STM32 on GitHub.
You can have access to these SVD directly on https://www.st.com.

You can access directly the SVD package under the category "CAD Resources", "System View Description" for each stm32 family .

Here is a link to have access to all the SVD file in one page: https://www.st.com/content/st_com/en/search.html#q=svd-t=resources-page=1

loicfrance, 2022-02-21:
Hello PierreLeCorre and CCASTM,
Thanks for the update. I understand that you cannot provide the svd files to us through github,
but is there any way we can have the mapping between devices and svd files ?
We can't really rely on the name, as there is a few exceptions.

suzizecat, 2022-02-21:
Hi @PierreLeCorre, To add on @loicfrance input, let' state our use case.
We are working on a system that would generate code targeting a given STM32 part (e.g. STM32F042K6). As this system requires the access to peripheral register definitions, we need few features:
 - The capability to automatically download the set of SVD files
 - The capacity to link a STM32 chip to its matching SVD (as it is sometimes non-trivial)

Moreover, the capacity to check for an update on said SVD files would be great as it could prevent us from downloading everything each time we need a generation.

ALABSTM, 2023-08-11:
Hi,

Unfortunately, no plan currently to address this request. Please allow me to close the thread. We will keep you informed shall there be any update. Thank you for your comprehension.

With regards,

closed 2023-08-11 by ALABSTM