STMicroelectronics/STM32Cube_MCU_Overall_Offer #4: CubeMX - Add code generation for PlatformIO

SunboX, 2021-02-08:
Hi,

as the subject states, it would be nice to have some pre-configured PlatformIO project "code" generation from CubeMX.

This is currently some barrier to start tinkering around with Nucleo boards. PlatformIO is used a lot in the Maker community and the STM32 support is really great atm. CubeMX generation PlatformIO projects would be even greater.

CCASTM, 2021-02-15:
Hello SunboX
The ST GitHub distribution is related to the Embedded Software publications. 
For any question or Issue related to CubeMX, I propose to post your question to the ST Community portal.
https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus
Regards
Christophe

CCASTM, 2021-02-24:
Hello SunBox,
If you don't mind, I propose to close this Issue. 
Regards
Christophe

closed 2021-07-06 by PierreLeCorre