STMicroelectronics/STM32Cube_MCU_Overall_Offer #5: examples procided

xRadx, 2021-02-16:
Hello STM devs,

I am not sure if this is the best place to write this issue/idea but the only I found.

Firstly thank yo for all the support, hoewer as an end user not much experienced yet, I found examples hard to use (in other words you idea is that your software provided should speed up "production process", however is not). With the Examples you do not provide .ioc file (which would be very useful for further adjustments), secondly the examples are not compatible with code generated by STMCube tools (MX or IDE) which again make things time consuming to make it working.
I also found Cube MX verry buggy (especially clock configurator).

CCASTM, 2021-02-19:
Hello xRadx,
Thank you for your positive feedback.
About the examples projects, we have started to provide the IOC format (CubeMX generated) for the last three years (STM32G0, G4, L5, WB, WL), and will be the case systematically in the future series 
Unfortunately, we did not retrofit this model to the legacy series as it would be a huge effort. 
For these legacy series, it is still possible to start a CubeMX project, select the same initialization settings than in the legacy example code and then integrate the rest of the code in the USER SECTIONs

About the clock generation, I propose to create a new post, with more details. But as it is a CubeMX related question, you should use the ST Forum https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus

Regards
Christophe

CCASTM, 2021-02-24:
Hello xRadx,

If you don't mind, I propose to close this Issue
Regards
Christophe

xRadx, 2021-03-05:
Let me use this issue for one more request.
I not able to run FATFS with USB host on nucleo h7a3zi (but I can do it on nucleo h743zi while following this guide: http://evenlund.blogspot.com/2016/10/usb-storage-with-stm32f4-discovery-and_58.html).
Step with crystal selection seems to have a big impact on getting it working on h743zi, while on H7a3zi nothing helps...
I noticed that function  USBH_UserProcess() is never called.

Would it be possible to add such application to the repo (including .ioc file)?

PierreLeCorre, 2021-06-08:
Hello @xRadx , sorry for the late notice.

This question is not related to this repository so I close the topic.

For the list of available example, you can refer to [AN5033: STM32Cube MCU Package examples for STM32H7 Series](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/6b/36/9b/ef/1d/91/4e/6b/DM00393275/files/DM00393275.pdf/jcr:content/translations/en.DM00393275.pdf) . The examples FatFs with USB are build on STM32H743I-EVAL. Unfortunately not on the board you are expected.

Don't hesitate to use this forum [stm32cubefw](https://community.st.com/s/topic/0TO0X0000003jBmWAI/stm32cubefw) to look for help to create your own application.

closed 2021-06-08 by PierreLeCorre