STMicroelectronics/STM32Cube_MCU_Overall_Offer #9: wrong repo name

tdjastrzebski, 2021-09-23:
Thank you for sharing HAL drivers etc. on Github!
I suggest rename this repo to something more meaningful.
1. Why call it `STM32Cube` when it is more about HAL?
1. Why call it `Overall Offer`? **Overall** implies something generic, non-specific while **offer** implies it is something for sale or trade exchange. Certainly this name does NOT encourage community participation. Quite opposite - to be honest.
It also makes this repo impossible to find if someone simply looks for STM32 HAL. 
IMHO it is a very bad name. I suggest get more opinions, consult it with someone more familiar with Github and English language usage.

PierreLeCorre, 2021-10-04:
Hi @tdjastrzebski,
the objective of this repository is to list **all** the **repositories** associated to the **STM32Cube** initiative.
You can find under **permissive license** for most of them. (Refer to the license of each repository):
- Firmware Package containing the Drivers, Middlewares, Utilities, Examples Project  for IAR Keil and STM32CubeIDE for a given STM32 serie
- X-Cube, STM32Cube expansion software containing embedded software components that complement the functionalities of the STM32Cube
- Board Support packages (BSP) containing the drivers of each STM boards
- Cmsis devices containing the register description of each STM32.
- Components drivers (mems, lcd, camera, audio etc) adapted for STM32Cube packages.
- And, as you mention, the HAL drivers of each STM32 series.

We have split our offer in two: **MCU** (described inside this repository) and MPU (inside https://github.com/STMicroelectronics/STM32MPU_EmbSW_Overall_Offer)

Then we are quite close of what you express.

tdjastrzebski, 2021-10-04:
Hi @PierreLeCorre, please just disregard it an close. You do not seem to understand my point I have no time to elaborate.

closed 2021-10-05 by PierreLeCorre