STMicroelectronics/STM32Cube_MCU_Overall_Offer #11: Issue with dfu prog on F767

rusefillc, 2021-12-06:
It looks like stm32cube programmer  does not erase proper page 12/erases non existent page 8 in some corner case of dual bank

See https://community.st.com/s/question/0D53W00001EEBXXSA5/stm32cubeprogrammer-incorrectly-erases-on-dual-bank-1mb-stm32f7

PierreLeCorre, 2022-01-03:
Dear @rusefillc , thanks to transmit this information. Unfortunately, this issue is not related to the content of this repository and so i closed it.

For your information, i have also transmitted your issue to our development team with the internal reference 120183.

closed 2022-01-03 by PierreLeCorre

rusefillc, 2022-01-03:
@PierreLeCorre is there a more relevant github repo for this issue? What's the best way to check on internal ticket 120183 going forward?

PierreLeCorre, 2022-01-03:
@rusefillc We don't provide access to our internal bug tracking system and we don't provide GitHub repository for the STM32CubeProgrammer where you can submit issues. Then, you have taken the correct path to submit your issue on  https://community.st.com.

rusefillc, 2022-04-11:
@PierreLeCorre this did not make into 2.10 update recently, do you know if there is hope for 2.11?

rusefillc, 2023-01-30:
fresh screenshot of 2.12

[image]

ALABSTM, 2023-08-11:
ST Internal Reference: 120183