STMicroelectronics/STM32Cube_MCU_Overall_Offer #19: Issue with dfu prog on F767 a year later

rusefillc, 2023-01-30:
Duplicate of #11

@PierreLeCorre can you please poke whoever is in charge of internal ticket 120183?

ALABSTM, 2023-08-11:
ST Internal Reference: 120183

ALABSTM, 2023-08-11:
Hi @rusefillc,

I sent a kind reminder to our development teams about this issue. I will get back to you as soon as I have their answer.

With regards,

ALABSTM, 2024-01-30:
Hi @rusefillc,

I hope you are doing well. Our development teams confirmed a fix has been implemented to fix the issue you reported. It will be part of release **2.16** of the [STM32CubeProgrammer](https://www.st.com/en/development-tools/stm32cubeprog.html), which should be available soon hopefully. We cannot share a precise date for the moment.

Please allow me to close this thread. Thank you for your comprehension and thank you again for having reported this issue.

With regards,

closed 2024-01-30 by ALABSTM