STMicroelectronics/STM32Cube_MCU_Overall_Offer #25: Improve Ease of changing MCU's

HarrisonTuckerTTIFC, 2023-08-16:
I have been using STM32 series for years and I tend to swap between the STM32G0, STM32F0, STM32F4, STM32G4 and now the STM32C0. I have been using drivers but would love to only have a general driver that worked for all STM32xx MCU's they all seem very similar and could be easily done with some macros( #ifdefs and such ) based on configurations of each chip.  this would allow validation to be easier since the same code would work for all MCUs and not need to be revalidated. also seems extensive to manage multiple sets of drivers that may get stale and have opputunity to improve the size and speed of code.

KORKAD, 2023-12-04:
Hello Harrison,
 
Thank you for this report :)
We can't manage overall STM32 series within only one generic driver for these main raisons:
1- The peripherals(IPs) designs aren't the same on overall STM32xx MCU's.
2- Same APIs available for same IPs features is already ensured==> No gain to reach on your application code
Even if we manage the overall diff under #ifdef you need to revalidate all concerned STM32xx MCU's (at least the part managed under #ifdef)
3- The HAL/LL drivers will be more complex and not easy to maintain even to understand them.
4- Major changes and high risk to break the API compatibility.
5- The ratio gain vs Effort&Cost not equitable
 
In addition, this request isn't aligned with 99% of Customers expectation: customer expect to have a simple/optimized HAL/LL drivers for only one target device.
 
So, I propose to close this request with status INVALID.
 
With Regards 
KORKAD

ALABSTM, 2024-03-25:
Hi @HarrisonTuckerTTIFC,

Thank you for this proposal. I hope the answer from @KORKAD clarified the point. Please allow me to close this thread. Thank you for your comprehension.

With regards,

closed 2024-03-25 by ALABSTM