STMicroelectronics/STM32Cube_MCU_Overall_Offer #26: X-CUBE-FREERTOS_V1.1.0 missing??? #1

r0oland, 2023-08-22:
This is a duplicate of https://github.com/STMicroelectronics/x-cube-freertos/issues/1 as I am not sure who should look into this...

Board: NUCLEO-U575ZI-Q
IDE: STM32CubeIDE Version: 1.13.1
Problem:

FreeRTOS examples are looking for "X-CUBE-FREERTOS_V1.1.0" which is nowhere to be found...

It can't be installed from anywhere and I can't create the project because it's missing.

[image]

HBOSTM, 2023-08-30:
Hello @r0oland ,

Thank you for this contribution, There is no problem from my side ,please re-install the CubeIDE and check the connection parameters if you are using a proxy.

Best Regards,

TOUNSTM, 2023-11-10:
Hi @r0oland,

I hope you are fine. Please allow me to close this issue as no reply has been received since a couple of months. Do not hesitate to reopen it in case you think the point is still relevant.

With regards,