STMicroelectronics/STM32Cube_MCU_Overall_Offer #36: STM32F767ZG GUI and command line display different (both wrong?) flash layout in dual mode

rusefillc, 2024-07-30:
A sub-set of https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/34 issue

command line tool calls rightly says that sector 0x08060000 is followed by 0x08080000 and calls it 38308 or sometimes 47462
```
erasing sector 0007 @: 0x08060000 done
erasing sector 38308 @: 0x08080000 done
Download in Progress:
```

while GUI calls it sector 12 while showing non-existed sector 11 probably related to https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/35

[image]

Worth noting that within same 2.17 CLI and GUI behaviors do not match.

rusefillc, 2024-07-30:
expected layout

[image]

ALABSTM, 2024-07-31:
Hi @rusefillc,

Thank you for this other report. However, as already said [here](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/11#issuecomment-1004105398), this support channel is mainly dedicated to provide support on firmware items published **within** [this](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer?tab=readme-ov-file#content) list of repositories.

Other support requests like yours, related to the ecosystem and tools, should be submitted to the **_ST Community_**, as already said [here](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/11#issuecomment-1004184039). Issues related to the **_STM32CubeProgrammer_** should be submitted in [this](https://community.st.com/t5/stm32cubeprogrammer-mcus/bd-p/stm32-mcu-cubeprogrammer-forum) forum.

Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2024-07-31 by ALABSTM

rusefillc, 2024-07-31:
@ALABSTM please consider reaching out to decision makers and relay the proposal for STM32CubeProgrammer to get a public issue tracker like this one here. Forums are just not suited for issue tracking unfortunatelly.