STMicroelectronics/STM32Cube_MCU_Overall_Offer #38: USB_OTG_HS missing from STM32H7R7A8Ix and STM32H7S7Z8Jx in CubeMX

doesnotcompete, 2024-09-21:
Hello,
while implementing [USB support for the H7R/S series within Embassy](https://github.com/embassy-rs/embassy/pull/3337) we have noticed that the CubeMX database is missing a definition for the `USB_OTG_HS` IP block. It is therefore also impossible to configure this peripheral within CubeMX, although both the ST website and CubeMX indicate those devices should also have HS USB with an integrated PHY.

ALABSTM, 2024-09-23:
ST Internal Reference: 191796

ALABSTM, 2024-09-23:
Hi @doesnotcompete,

Thank you for the heads-up. Your report has been logged internally to be considered for future versions of the `STM32CubeMX` tool.

With regards,

ALABSTM, 2024-10-03:
Hi @doesnotcompete,

Our hardware design team confirmed none of `STM32H7R7A8Ix` and `STM32H7S7Z8Jx` part numbers implement `USB OTG HS`. It looks like the issue is rather due to what is indicated on [st.com](https://www.st.com/en/microcontrollers-microprocessors/stm32h7r7a8.html).

With regards,

ALABSTM, 2024-10-14:
Hi @doesnotcompete,

I hope you are doing well. As already communicated, none of `STM32H7R7A8Ix` and `STM32H7S7Z8Jx` part numbers implement `USB OTG HS`, only `USB OTG FS`.

The information available on _st.com_ has been updated. Please allow me to close this issue. Our apology for any inconvenience. Thank you for your comprehension.

With regards,

closed 2024-10-14 by ALABSTM