STMicroelectronics/STM32Cube_MCU_Overall_Offer #46: No examples for OpenBootloader in F4 MCU package

jp54l, 2025-01-31:
The OpenBootloader usage / configuration is difficult to follow and configure.

Firstly, there is OpenBootloader supplied under any of the projects or boards for the STMF4 in the Firmware Package.

Two, there is no configuration or downloadable package eg. X-CUBE-OPENBL like other middleware,

This feature is very important in situations where a new firmware image is to be supplied OTA, and where the System Bootloader cannot be processed.

I have successfully rolled my own, OTA firmware update over Bluetooth that first stores the image to external flash, resets the MCU, then reads out the firmware image from external flash and writes to internal memory.

The biggest hurdle that I am yet to overcome, is how to write to unlock flash when read protected and run the bootloader from RAM, this is not so easy.

Anyway, my main point is that I would prefer to use a standardized STM supplied bootloader, but there are no examples for any F4 mcus projects nor is it included in the MCU Firmware package.

What do you suggest as a way forward, or better yet can this be resolved, as I would think would be a straightforward for STM to fix.

ALABSTM, 2025-02-05:
Hi @jp54l,

I will ask whether such examples are planned for the STM32F4 MCU. In the meanwhile, you can have a look at the following repositories, listed in [this](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer?tab=readme-ov-file#stm32cube-middleware-libraries-and-applications) section of the README.md:
* [stm32l5-openbl-apps](https://github.com/STMicroelectronics/stm32l5-openbl-apps)
* [stm32wb-openbl-apps](https://github.com/STMicroelectronics/stm32wb-openbl-apps)
* [stm32wl-openbl-apps](https://github.com/STMicroelectronics/stm32wl-openbl-apps)

With regards,

ALABSTM, 2025-02-05:
Hi @jp54l,

Unfortunately, there is currently no plan to release examples showing the use of the open bootloader library with a STM32F4 MCU. I hope the examples I pointed out will help you. You can find further help from the **ST Community** [here](https://community.st.com/t5/stm32-mcus/ct-p/stm32-mcus). [This](https://community.st.com/t5/stm32-mcus/how-to-use-the-st-open-bootloader-for-stm32-microcontrollers/ta-p/49896) article may also be useful.

Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2025-02-05 by ALABSTM